YukMingLaw/ArduinoJoystickWithFFBLibrary #2: Englisch and Pinout

gagagu, 2020-05-12:
Hi,
firstable thanks for your work and sharing it with us!
I want to use your library for a force feedback flight stick but i don't know how to connect the electric components. 
Is it possible for you to make a readme in englisch so that everyone in the world can read it?

Thx forward!

YukMingLaw, 2020-05-12:
Hi,the english readme is added now.But the library is not support multi-forcefeedback-axis now.The flight stick needs two axis with forcefeedback usually. I will add the multi-forcefeedback-axis feature in the next version.

YukMingLaw, 2020-05-12:
Can you tell me which game or simulator program you test？I will test the library on it.

gagagu, 2020-05-12:
I am using X-Plane 11 atm, MS 2020 later when ist out

YukMingLaw, 2020-05-17:
The double-axis-force-feedback feature is added.But it seems that the X-Plane 11 does not have force feedback?You can test the library in your machine.

closed 2020-05-17 by YukMingLaw

pkaser, 2020-05-17:
Thank you so very much! I too will be using this in X-Plane I'll let you know how I get it to work

pkaser, 2020-05-19:
Hey bud, thanks for writing this. I was wondering, is there any kind of wiring diagram to go along with this?