YukMingLaw/ArduinoJoystickWithFFBLibrary #3: JOYSTICK_TYPE_GAMEPAD  Crashes DCS (Digital Combat Simulator)

omarachrafbrasil, 2020-05-19:
The sketch below works in windows joystick usb configuration. However when I start DCS (Digital Combat Simulator) DCS crashes.

/*
	switch in pin 2
	potentiometers in A0, A1 and A2
	Arduino pro micro
*/

#include "Joystick.h"

Joystick_ Joystick(
  JOYSTICK_DEFAULT_REPORT_ID, // hidReportId
  JOYSTICK_TYPE_GAMEPAD,      // joystickType
  1,   // buttonCount
  0,    // hatSwitchCount
  true, // includeXAxis
  true, // includeYAxis
  true, // includeZAxis
  false,// includeRxAxis
  false,// includeRyAxis
  false,// includeRzAxis
  false,// includeRudder
  false,// includeThrottle
  false,// includeAccelerator
  false,// includeBrake 
  false // includeSteerings
 );

const int pinButton = 2;

int sx, sy, sz; // sensor values
int x, y, z; // values to Joy X, Y, Z

void setup() {
  pinMode(pinButton, INPUT_PULLUP);
    
  Joystick.setXAxisRange(-1023, 1023);
  Joystick.setYAxisRange(-1023, 1023);
  Joystick.setZAxisRange(-127, 127);
  
  Joystick.begin(false);
}

void loop() {
  if (digitalRead(pinButton)) {
    Joystick.pressButton(0); // 0..31
  } else {
    Joystick.releaseButton(0);
  }

  sx = analogRead(A0);
  sy = analogRead(A1);
  sz = analogRead(A2);

  x = map(sx, 0, 1023, -1023, 1023);
  y = map(sy, 0, 1023, -1023, 1023);
  z = map(sz, 0, 1023, -127, 127);

  Joystick.setXAxis(x);
  Joystick.setYAxis(y);
  Joystick.setZAxis(z);
  
  Joystick.sendState();

  delay(20);
}

YukMingLaw, 2020-05-19:
Hi,I compiled your code and uploaded it to my arduino pro micro.It works nice.I cant recurrent the problem what you said.Maybe change another chip and try again?

omarachrafbrasil, 2020-05-19:
adasffsdafsdafsdaf

omarachrafbrasil, 2020-05-19:
Hi, had you tried running DCS after your pro micro is plugged?
It worked for me too. Even it is possoble to perform the Settings in the Windows Game Controllers Windows.
The problem happens when the DCS (Digital combat simulator) is executed. DCS crashes with pro micro running this sketch/in my board.
If there my pro micro with is not plugged and DCS is executed, DCS dont crash.

YukMingLaw, 2020-05-19:
You can delete `delay(20);` and try again.Or use more less delay like `delay(1);`.

omarachrafbrasil, 2020-05-19:
Hi, I commented out delay(20) and tried again. Then run DCS. Still crashed.
In a DCS forum of 2 years ago someone commented regarding FFD on gamepads.
This promote some Idea about the problem?
Omar Achraf

omarachrafbrasil, 2020-05-19:
Hi, just another feedback. I tested with the library that doesnt have FFB (
https://github.com/MHeironimus/ArduinoJoystickLibrary ) and it worked ok
with DCS.  Could be related to FFB?
Omar Achraf

YukMingLaw, 2020-05-20:
got it,I will test the library in DCS soon:)

YukMingLaw, 2020-05-23:
Hi,I have recurrented this problem yet.You must add `Joystick.getForce(forces);` to your code.
Because the force feddback part is appended to the HID descriptor,so when the game start,it will ask for the force feedback reply from device.

closed 2020-05-25 by YukMingLaw

YukMingLaw, 2020-05-25:
I updated the library.It changed a lot.You should download the latest version of library to make your program.

omarachrafbrasil, 2020-05-26:
Hi,
Updated! and recompiled and upload to my pro micro (without
Joystick.getForce(forces);  )

It executed DCS simulator withou crash.
But, when I  exit DCS, DCS hangs. Need too kill.

Omar Achraf