YukMingLaw/ArduinoJoystickWithFFBLibrary #8: Recreate Logitech WingMan Force with Arduino Leonardo

CryptoAngel, 2020-06-28:
Hello,

I have Logitech WingMan Force for very long time, but on Windows 10 it is not supported anymore (no driver) so I decided to replace the board with Arduino Leonardo. I already dismantled the board and connected the Leonardo to the potentiometers with the simple joystick x,y sample.
But I have lots of buttons and also hat buttons and most important I have two motors for "force back" that I would love to have them work as in the old days ;-)

Can you help with code for that?
Also I'll be happy for sketch how connect wires to Leonardo legs.

Of course I will upload this project with video for the benefit of everyone with credits ;-D

Thank you so much in advance

storca, 2020-06-28:
Hello

I can't help you personnaly with code because I think that you have to code it yourself.

However, in a project like this, if I was you and if you haven't done it yet, you should ask support or community for a W10 driver or for a trick to install drivers on W10.
If you don't find any way of installing old drivers, then maybe you should consider replacing the board.

If you are looking for code, you should have a look in the examples folder. _I'm not sure that there is an example with FFB, I'll make a PR if that helps_ 

If you have joystick / buttons working, then you might take a look on how to control the motors. 

Hope this helps

storca, 2020-06-28:
Found this while searching for drivers

https://www.userdrivers.com/Keyboard-Mouse/Logitech-Gaming-Software-5-10-127-for-Windows-XP-Vista-7-64-bit/

closed 2020-06-28 by CryptoAngel

storca, 2020-06-30:
Did that work for you or you still need example ffb code?