YukMingLaw/ArduinoJoystickWithFFBLibrary #9: Example file missing

idrislaode, 2020-07-13:
Hi. I cant find example joystickWithFFB file? its removed?

YukMingLaw, 2020-07-28:
Hi,
The example code was removed.
You can refer to readme as a example code.

waggy685, 2020-07-28:
Same for me.

YukMingLaw, 2020-07-28:
Hi,example code is added.[https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary/tree/master/examples/SingleJoystickFFB](url)

Scatman96, 2020-07-28:
Hi,
how to interface the example code with a DIY Yoke/Joystick  or  stepper/DC-motors/BLDC`s with encoders ?

i hope its "easy" to do. 

Can i use the   "forces[0] is the X-Axis force data "   "forces[1] is the Y-Axis force data"  directly for the motor ? or do i need some "math" ?

thanks you very much for your great Library

YukMingLaw, 2020-07-28:
Hi,
The `forces[0]` is a data in range[-255,255].So if you use a DC-motor,just use it as a PWM value,Ofcourse use absolute value.
here is a sample:
```
if(forces[0] > 0){
  digitalWrite(4,HIGH);
  digitalWrite(5,LOW);
  analogWrite(7,abs(forces[0]));
}else{
  digitalWrite(4,LOW);
  digitalWrite(5,HIGH);
  analogWrite(7,abs(forces[0]));
}
```
If you use a BLDC-Motor or a steper-Motor,you should write some code to drive it.And bind this value `forces[0]` as the motor strength.
Hopes it can help you.

Scatman96, 2020-07-28:
ok thank you very much :) 

So I understand that the encoders are on the output side of the joystick and have nothing to do with the force feedback? only the position of the joystick? Right ? :D

storca, 2020-07-28:
HelloI think that there are some mistakes in the example code.I'll take a look in 2-3hrs and make a pull request.

storca, 2020-07-28:
Hello

I think that there are some mistakes in the example code, I'll take a look in 2-3hrs and make a pull request.

Scatman96, 2020-07-28:
i also have problems compiling

_SingleJoystickFFB:15:1: error: 'mygains' does not name a type
 mygains[0].springGain = 100;//0-100_

idrislaode, 2020-07-29:
how to set ffb for steering wheel?

idrislaode, 2020-07-29:
i get error compiling

SingleJoystickFFB:14:1: error: 'mygains' does not name a type
 mygains[0].totalGain = 100;//0-100
 ^~~~~~~
SingleJoystickFFB:15:1: error: 'mygains' does not name a type
 mygains[0].springGain = 100;//0-100
 ^~~~~~~
C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_694105\SingleJoystickFFB.ino: In function 'void loop()':
SingleJoystickFFB:41:21: error: 'forces' was not declared in this scope
   Joystick.getForce(forces);
                     ^~~~~~
C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_694105\SingleJoystickFFB.ino:41:21: note: suggested alternative: 'fopen'
   Joystick.getForce(forces);
                     ^~~~~~
                     fopen
Using library ArduinoJoystickWithFFBLibrary-master at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master 
exit status 1
'mygains' does not name a type

YukMingLaw, 2020-07-29:
Hi,All.
The new example code is available and successful in compiling.

Scatman96, 2020-07-29:
> Hi,All.
> The new example code is available and successful in compiling.

Hi,
could you comment on the codes so everyone knows what's going on there?

```
    pinMode(A2,INPUT);
    pinMode(9,OUTPUT);
    pinMode(6,OUTPUT);
    pinMode(7,OUTPUT);
```

`  int value = analogRead(A2);`

thank you very much for your fast correction of the example sketch

closed 2020-10-01 by YukMingLaw