YukMingLaw/ArduinoJoystickWithFFBLibrary #10: Gains and Effects?

gagagu, 2020-07-28:
Hi,
i've used the based joystick library to repair my broken Saitek Yoke and it's working perfect. Now, i want to extend it to a FFB Yoke. My plan is to use a strong RC Servo (35Kg) to create the force. The Serve should rotate in the same angle as the yoke plus or minus the measured force.
I've made some tests with X-Plane and the XPForce Plugin but i have some problems to understand the Gains and Effects Parameters. Basely i do understand for what they are but i don't have enough experience to select the correct gains and parameters settings.
Is it possible for you to help me out to find the correct configuration?

Thx forward for your work!

YukMingLaw, 2020-07-28:
Hi,
The gains are the magnitude of each force,range0-100.
The effects parameters are the dynamic parameters,they should bind with the current state of the action-axis.Like axis's position(degree),axis's speed(delta degree),and axis's accelation(delta speed).
Hopes it can help you.

waggy685, 2020-07-28:
> Hi,
> i've used the based joystick library to repair my broken Saitek Yoke and it's working perfect. Now, i want to extend it to a FFB Yoke. My plan is to use a strong RC Servo (35Kg) to create the [...]

Hi GAGAGU !

Would you share your *.ino file ? Because a got stuck... I have the same Joystick. Thanks !

Best regards

gagagu, 2020-07-28:
@YukMingLaw:
I't helps me a little bit, even if i don't know which Gains i should use.

@waggy685:
Yes, of course. If you wonder what i've done: 
1: i didn't know the map command when programming this.
2: I've made a small Filter routine to flatten the reaction curve

If you have some problems with my german langage, contact me for help.

#include "Joystick.h"

#define JOYSTICK_THROTTLE_ID         0x03

Joystick_ Joystick(JOYSTICK_THROTTLE_ID,
                   JOYSTICK_TYPE_JOYSTICK, 14, 1,
                   true, true, false, false, false, false,
                   false, false, false, false, false);

// Number of Readings for Smoothing
const int numReadings = 20;
int readIndexDigitalPins = 0;              // the index of the current reading

// Constant that maps the phyical pin to the joystick button.
int inputPinQuerruder = A0;
int inputPinHoehenruder = A1;

// Last state of the button
int lastButtonState[14] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
int val0 = 0;
int val1 = 0;

int alteWerte[3] = {0, 0};
float diff = 0.0;
float teiler = 0.0;
float valFloat = 0.0;

// Höhenruder
int hoeheUnderBorder = 315;
int hoeheUpperBorder = 900;

int readingsHoehenruder[numReadings];      // the readings from the analog input
int readIndexHoehenruder = 0;              // the index of the current reading
int totalHoehenruder = 0;                  // the running total
int averageHoehenruder = 0;                // the average

//  Querruder
int querUnderBorder = 330;
int querUpperBorder = 920;

int readingsQuerruder[numReadings];      // the readings from the analog input
int readIndexQuerruder = 0;              // the index of the current reading
int totalQuerruder = 0;                  // the running total
int averageQuerruder = 0;                // the average

void setup() {
  Joystick.setXAxisRange(querUnderBorder, querUpperBorder);
  Joystick.setYAxisRange(hoeheUnderBorder, hoeheUpperBorder);

  pinMode(A0, INPUT_PULLUP);
  pinMode(A1, INPUT_PULLUP);
  pinMode(A2, INPUT_PULLUP);
  pinMode(A3, INPUT_PULLUP);
  pinMode(A4, INPUT_PULLUP);
  pinMode(A5, INPUT_PULLUP);

  pinMode(0, INPUT_PULLUP);
  pinMode(1, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
[... 116 lines cut ...]
  averageQuerruder = totalQuerruder / numReadings;
    
  if (readIndexQuerruder >= numReadings) {
    readIndexQuerruder = 0;
    Joystick.setXAxis(averageQuerruder);
  }

  // ************************************************************************************************
  //  Hoehenruder
  // ************************************************************************************************

  totalHoehenruder = totalHoehenruder - readingsHoehenruder[readIndexHoehenruder];
  readingsHoehenruder[readIndexHoehenruder] = analogRead(inputPinHoehenruder);
  delay(1);
  totalHoehenruder = totalHoehenruder + readingsHoehenruder[readIndexHoehenruder];
  readIndexHoehenruder= readIndexHoehenruder + 1;
  averageHoehenruder = totalHoehenruder / numReadings;
    
  if (readIndexHoehenruder >= numReadings) {
    readIndexHoehenruder = 0;
    Joystick.setYAxis(averageHoehenruder);
  }

  //delay(5);
}

YukMingLaw, 2020-07-28:
Hi,
#define FORCE_FEEDBACK_MAXGAIN              100
struct Gains{
        uint8_t totalGain         = FORCE_FEEDBACK_MAXGAIN;
	uint8_t constantGain      = FORCE_FEEDBACK_MAXGAIN;
	uint8_t rampGain          = FORCE_FEEDBACK_MAXGAIN;
	uint8_t squareGain        = FORCE_FEEDBACK_MAXGAIN;
	uint8_t sineGain          = FORCE_FEEDBACK_MAXGAIN;
	uint8_t triangleGain      = FORCE_FEEDBACK_MAXGAIN;
	uint8_t sawtoothdownGain  = FORCE_FEEDBACK_MAXGAIN;
	uint8_t sawtoothupGain    = FORCE_FEEDBACK_MAXGAIN;
	uint8_t springGain        = FORCE_FEEDBACK_MAXGAIN;
	uint8_t damperGain        = FORCE_FEEDBACK_MAXGAIN;
	uint8_t inertiaGain       = FORCE_FEEDBACK_MAXGAIN;
	uint8_t frictionGain      = FORCE_FEEDBACK_MAXGAIN;
	uint8_t customGain        = FORCE_FEEDBACK_MAXGAIN;
};
The struct initialized with max gain,so if you dont know how to set this param.just creat a gain var `Gains mygains[2];` and invoke `setGains(mygains)`.

closed 2020-07-28 by YukMingLaw