YukMingLaw/ArduinoJoystickWithFFBLibrary #11: will not show as controller

jwesthoff04, 2020-07-29:
will compile but wont show up as a controller. if i use the ArduinoJoystick library without the FFB it shows as controller. also force tester wont recognize it as a controller unless i use the library without FFB it shows but since no FFB it cant test anything. please help. im trying to use with Assetto corsa. 

This is the code for no ffb if i use your library it dose not work if i use the other it dose.

#include <Joystick.h>

// set joystick to library
// always start id at 0x03 also defalt
// 0x01,and 0x02 are used for internal mouse and keyboard lybrary
Joystick_ Joystick(0x03, JOYSTICK_TYPE_MULTI_AXIS, 8, 0, false, false, false, false, false, false, false, true, true, true, true);

const int steeringpot = A0;
const int throttlepot = A1;
const int brakepot = A2;
const int clutchpot = A3;

//arrys for buttons and sets state to zero
const int button[8] = { 5, 6, 7, 8, 9, 10, 11, 12};
int lastButtonState[8] = {0, 0, 0, 0, 0, 0, 0, 0};

void setup() {
  //Serial.begin(9600);
  // sets joystick
  Joystick.begin();

  // sets pins to input
  for (int i = 0; i < 8; i++) {
    pinMode(button[i], INPUT_PULLUP);
  }

}

void loop() {

  // reads pots and sends data to control steering, throttle, brake, and clutch
  // delay to read pots
  float steering = analogRead(steeringpot);
  Joystick.setSteering(steering);
  delay(1);

  float throttle = analogRead(throttlepot);
  throttle = map(throttle, 391.00, 750.00, 0.00, 1023.00);
  Joystick.setThrottle(throttle);
  delay(1);

  float brake = analogRead(brakepot);
  Joystick.setBrake(brake);
  delay(1);

  float clutch = analogRead(clutchpot);
  Joystick.setAccelerator(clutch);

  for (int i = 0; i < 8; i++) {
    // Read pin values and replaces lastButtonState with new state
    int currentButtonState = !digitalRead(button[i]);
    if (currentButtonState != lastButtonState[i])
    {
      Joystick.setButton(i, currentButtonState);
      lastButtonState[i] = currentButtonState;
    }

  }
  //Serial.println(throttle);
}

This is the code for ffb and it compiles and shows as a controller in my devices but will not work on anything. please help!!!

#include <Joystick.h>
#include <Encoder.h>

int32_t forces[2] = { 0 };

long posX;
long posY;

int centerX;
int centerY;

int lastX = 0;
int lastY = 0;

// Motor
int dirX = 12;
int dirY = 13;

int brakeX = 9; // NIU
int brakeY = 8; // NIU

int motorX = 3; //
int motorY = 11; //

//current draw on motors
int loadX = A0;
int loadY = A1;

// joystick encoders
int encoderX1 = 0;
int encoderX2 = 1;
int encoderY1 = 2;
int encoderY2 = 4;

int xDirection = LOW;
int yDirection = LOW;
int centerFound = LOW;

int dur = 2;
int strength = 100;

long oldX = -999;
long oldY = -999;

const int loopDelay = 20;

int minX = 0;
int maxX = 0;
int minY = 0;
int maxY = 0;
const int throttlepot = A3;
const int brakepot = A2;
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK,
0, 0, // Button Count, Hat Switch Count
true, true, false, // X and Y, but no Z Axis
false, false, false, // No Rx, Ry, or Rz
false, true, // No rudder or throttle
true, true, true); // No accelerator, brake, or steering

Gains gain[2];
EffectParams effects[2];

//

Encoder encoderX(encoderX1, encoderX2);
Encoder encoderY(encoderY1, encoderY2);

void setup() {
Serial.begin(9600);
delay(2000); //Give the serial port time to catch up so we can debug
Serial.println("Setting up...");
pinMode(dirX, OUTPUT); //Initiates Motor Channel X pin
pinMode(dirY, OUTPUT); //Initiates Motor Channel Y pin

Joystick.begin(true);
doSetGains();
}

void loop(){
  float throttle = analogRead(throttlepot);
  throttle = map(throttle, 391.00, 750.00, 0.00, 1023.00);
  Joystick.setThrottle(throttle);
  delay(1);

  float brake = analogRead(brakepot);
  Joystick.setBrake(brake);
  delay(1);
  
if (centerFound == 0){
Serial.println("--Centering");
doFindCenter();
}
doJoystickStuff();
doFFBStuff();
delay(loopDelay);
}

void doFindCenter(){
int cDur = 80;
int cStrength = 255;
int tXDir;
int tYDir;
int tPosX;
int tPosY;

doMoveX(cStrength,cDur,HIGH);
maxX = stopXMove();
doMoveX(cStrength,cDur,LOW);
minX = stopXMove();
int centerX = (minX + maxX)/2;
doMoveY(cStrength,cDur,HIGH);
maxY = stopYMove();
doMoveY(cStrength,cDur,LOW);
minY = stopYMove();
int centerY = (minY + maxY)/2;

centerX = ((minX + maxX)/2)-2;
centerY = ((minY + maxY)/2)-2;

Joystick.setXAxisRange(minX,maxX);
Joystick.setYAxisRange(minY,maxY);
tPosX = encoderX.read();
tPosY = encoderY.read();
do {
//center X
if (centerX > tPosX){tXDir = HIGH;} else {tXDir = LOW;}
doMoveX(255,2,tXDir);
tPosX = stopXMove();

} 
while (tPosX != centerX);
do {
//center Y
if (centerY > tPosY){tYDir = HIGH;} else {tYDir = LOW;}
doMoveY(255,2,tYDir);
tPosY = stopYMove();

} 
while (tPosY != centerY);
stopXMove();
stopYMove();
centerFound = 1;
}

int stopXMove(){
return doMoveX(0, 10, HIGH);
}

int stopYMove(){
return doMoveY(0, 10, HIGH);
}

  
  

int doMoveX(int tStrength, int tDuration, int tDirection){
digitalWrite(dirX, tDirection); //High=Forward / Low=Reverse
analogWrite(motorX, tStrength); //Spins the motor on Channel A at full speed
delay(tDuration);
// analogWrite(motorX, 0); //Stops motor
delay(2);
return encoderX.read();
}

int doMoveY(int tStrength, int tDuration, int tDirection){
digitalWrite(dirY, tDirection); //High=Forward / Low=Reverse
analogWrite(motorY, tStrength); //Spins the motor on Channel A at full speed
delay(tDuration);
// analogWrite(motorY, 0); //Stops motor
delay(2);
return encoderY.read();
}

void doFFBStuff(){
//}

}

void doJoystickStuff(){

posX = encoderX.read();
posY = encoderY.read();

Joystick.setXAxis(posX);
Joystick.setYAxis(posY);

effects[0].springMaxPosition = maxX/2;
effects[1].springMaxPosition = maxY/2;
effects[0].frictionMaxPositionChange = lastX - posX;
effects[1].frictionMaxPositionChange = maxY;
effects[0].inertiaMaxAcceleration = maxX;
effects[1].inertiaMaxAcceleration = 100;
effects[0].damperMaxVelocity = 10;
effects[1].damperMaxVelocity = 100;

effects[0].springPosition = posX;
effects[1].springPosition = posY;
effects[0].frictionPositionChange = lastX - posX;
effects[1].frictionPositionChange = lastY - posY;
effects[0].inertiaAcceleration = 100;
effects[1].inertiaAcceleration = 100;
effects[0].damperVelocity=100;
effects[1].damperVelocity=100;

Joystick.setEffectParams(effects);
Joystick.getForce(forces);

int xMoveDir = 0;
int yMoveDir = 0;
//Get Force [-255,255] you can set PWM with this value
Serial.println("");
Serial.print(" - XF: ");
Serial.print(forces[0]);
Serial.print(" YF: ");
Serial.print(forces[1]);
//if (forces[0] != 0 || forces[1] != 0 ){
if (forces[0]<0){
xMoveDir = LOW;
} else if (forces[0]>0){
xMoveDir = HIGH;
}
if (forces[1]<0){
yMoveDir = LOW;
} else if (forces[1]>0){
yMoveDir = HIGH;
}
if (forces[0]!=0){
doMoveX(abs(forces[0]),dur,xMoveDir);
} else {
doMoveX(0,2,xMoveDir);
};
if (forces[1]!=0) {
doMoveY(abs(forces[1]),dur,yMoveDir);
} else {
doMoveY(0,2,yMoveDir);
};

lastX = posX;
lastY = posY;

}

void doSetGains(){
//set x axis gains
gain[0].totalGain = 100;
gain[0].constantGain = 100;
gain[0].rampGain = 100;
gain[0].squareGain = 100;
gain[0].sineGain = 100;
gain[0].triangleGain = 100;
gain[0].sawtoothdownGain = 100;
gain[0].sawtoothupGain = 100;
gain[0].springGain = 100;
gain[0].damperGain = 100;
gain[0].inertiaGain = 100;
gain[0].frictionGain = 100;

//set y axis gains
gain[1].totalGain = 100;
gain[1].constantGain = 100;
gain[1].rampGain = 100;
gain[1].squareGain = 100;
gain[1].sineGain = 100;
gain[1].triangleGain = 100;
gain[1].sawtoothdownGain = 100;
gain[1].sawtoothupGain = 100;
gain[1].springGain = 100;
gain[1].damperGain = 100;
gain[1].inertiaGain = 100;
gain[1].frictionGain = 100;

Joystick.setGains(gain);

}

YukMingLaw, 2020-07-29:
Hi,
you can try this constructed function.Maybe there are some conflictions between accelerator, brake, steering and FFB.
```
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK,
  8, 0,                  // Button Count, Hat Switch Count
  true, true, true,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering
```

YukMingLaw, 2020-07-29:
And...I found that you use two encoders.Can you explane their's usage?

jwesthoff04, 2020-07-29:
no only need one for steering and i tried setting as you said and still not working.

YukMingLaw, 2020-07-29:
Do you set X-axis as the steering in AC?
My setting:
X-axis-->steering
Y-axis-->break
Z-axis-->accelerator

jwesthoff04, 2020-07-29:
got it working you can not use anything but x,y,  or z. thank you so much!!!!!!

freppac, 2020-11-13:
I have the same issue...running the singlejoystickffb unchanged on a arduino micro. It does not show up as a game controller in w10. Do I need a driver installed?

aeroelectrolight, 2020-11-25:
I have the same issue too .... But if i test it with fedit, he recognizes it as a force feedback device and aame with wheelcheck ! I dont understand. Can you help us please ?

freppac, 2020-11-25:
Actually I gor it working....do not use JOYSTICK_TYPE_MULTI_AXIS in the instantiation...I used JOYSTICK_TYPE_JOYSTICK and it worked!

aeroelectrolight, 2020-12-11:
Thank you very much, it work for me to !

positron96, 2021-12-03:
Can the example be changed then? `examples/SingleJoystickFFB/SingleJoystickFFB.ino` has erroneous JOYSTICK_TYPE_MULTI_AXIS which is not recognized correctly by Windows.