YukMingLaw/ArduinoJoystickWithFFBLibrary #12: Support for stm32

joesphan, 2020-08-13:
Hello! How easy would this be to port to stm32duino? I saw the mmos firmware so I pulled the trigger, but it looks like it doesnt have the source code. I'm willing to work with you to get this done.

YukMingLaw, 2020-08-13:
Hello,I have not used stm32duino yet,so I have to evaluate it first.You are welcome to contribute to this project!

joesphan, 2020-08-14:
After much research, I am to find that I am not qualified to take on this project. Through my research though, I found out that this library indeed has the right HID PID descriptors for FFB operation, and should be easy to integrate into any MCU with native usb. I ended up purchasing the arduino micro and going with @YukMingLaw 's source code instead.

closed 2020-08-14 by joesphan