YukMingLaw/ArduinoJoystickWithFFBLibrary #13: higher than 8 bit force feedback resolution

joesphan, 2020-09-10:
Hello! Sorry to bug you again, I have implemented your code into my FFB wheel project (on my page), using a mige servo and a generic servo drive. I am wondering if it is possible to get higher than 9 bit resolution on the force feedback (8 bit each way, -255 to 255). The game I have tested is BeamNG.drive, and it only gives me 8 bit values. I would like to take advantage of the 13 bit analog control resolution on my servo drive. 

Thanks in advance!

lols11, 2020-09-10:
In file Joystick.cpp (line 581 and 582) in function which returning force there is code:  `forces[0] = constrain(forces[0], -255, 255);
	forces[1] = constrain(forces[1], -255, 255);`
You can change the range. But I am not sure if the higher values ​​will be correctly calculated.

YukMingLaw, 2020-09-10:
Hi，I set the value mapping in:
https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary/blob/45ee061d4e71ee2fea0028fa0f76bbae1a2398cf/src/Joystick.cpp#L581
you can modify the mapping value range yourself.

joesphan, 2020-09-10:
Thanks for your reply. I made a fork right now implementing an adjustable range, feel free to merge it.
`Joystick.cpp:`
```cpp
/*  J.L 9/10/2020
*   modified ffb range to depend on the variable 'FFBRange'
*/

void Joystick_::forceCalculator(int32_t* forces) {
///////////////////////////////////
	forces[0] = constrain(forces[0], -_FFBRange, _FFBRange);
	forces[1] = constrain(forces[1], -_FFBRange, _FFBRange);
}
```
`Joystick.h:`
```cpp
/*  J.L 9/10/2020
*   range of force feedback, default is 9 bit (-255 to 255)
*   can be set using joystick::setFFBRange(int range)
*/
int _FFBRange = 255;  
///////////////////////////////////
void Joystick_::setFFBRange(int range)
{
    _FFBRange = range;
}
////////////////////////////////////
```
the README in my fork is updated to match as well. calling of gains is different from before as well.

thanks for all your help with this project of mine, without your library, I would still be stuck on getting USB HID to work.

closed 2020-10-01 by YukMingLaw