YukMingLaw/ArduinoJoystickWithFFBLibrary #23: FFB support for Arduino DUE (32bit ARM)

NCPlyn, 2021-02-21:
Firstly, this library is amazing and I'm happy that I found it. 
It works nicely on 32u4, but that this chip is kinda too slow. My code takes like **9ms** to execute. (Set axis and buttons). 
For example, Forza Horizon 4, the frame rate drops to 4FPS because it's waiting for the chip to get data. (Physics refresh at 300hz) Assetto corsa won't even start the race. 
So I realized that the library that this is build on supports Arduino DUE that's multiple times faster (now the execution time is only **0.85ms**).

Now comes the problem with this library, I can see in the code a few tries to get it working on DUE, but only something...
In the DynamicHID.cpp the USB_Send is correctly translated to USB**D**_Send, but there are multiple other than this.
I tried to fix it and I got no errors regarding that, but there might be some problems because the DUE core has uint32 in the place instead of the uint8 in USBCore.cpp, that might couse problems...

Also I'm getting this error:

```
C:\Users\Speedy\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master\src\DynamicHID\DynamicHID.cpp: In constructor 'DynamicHID_::DynamicHID_()':
C:\Users\Speedy\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master\src\DynamicHID\DynamicHID.cpp:242:65: error: no matching function for call to 'PluggableUSBModule::PluggableUSBModule(int, int, uint8_t [2])'
                    protocol(DYNAMIC_HID_REPORT_PROTOCOL), idle(1)
                                                                 ^
C:\Users\Speedy\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master\src\DynamicHID\DynamicHID.cpp:242:65: note: candidates are:
In file included from C:\Users\Speedy\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master\src\DynamicHID\DynamicHID.h:35:0,
                 from C:\Users\Speedy\Documents\Arduino\libraries\ArduinoJoystickWithFFBLibrary-master\src\DynamicHID\DynamicHID.cpp:22:
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:29:3: note: PluggableUSBModule::PluggableUSBModule(uint8_t, uint8_t, uint32_t*)
   PluggableUSBModule(uint8_t numEps, uint8_t numIfs, uint32_t *epType) :
   ^
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:29:3: note:   no known conversion for argument 3 from 'uint8_t [2] {aka unsigned char [2]}' to 'uint32_t* {aka long unsigned int*}'
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:27:7: note: constexpr PluggableUSBModule::PluggableUSBModule(const PluggableUSBModule&)
 class PluggableUSBModule {
       ^
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:27:7: note:   candidate expects 1 argument, 3 provided
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:27:7: note: constexpr PluggableUSBModule::PluggableUSBModule(PluggableUSBModule&&)
C:\Users\Speedy\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/USB/PluggableUSB.h:27:7: note:   candidate expects 1 argument, 3 provided
```

Which shows the problem that the SAM (DUE) core expects uint32, not uint8 (tried to fix that but when app tries to access the FFB data, it crashes) and also I've found that the USBAPI.h from the SAM core is missing some things that are connected with the "PluggableUSBModule", for example "EP_TYPE_INTERRUPT_IN".

I'm not very good at this level of programming, but I hope that I pointed out some things and I will be happy if somebody can help me port this library to the way faster Arduino DUE.

NCPlyn, 2021-02-22:
OK, so I completely forgot to check pull requests. There is fix for due boards by @billydragon , but after applying the changes, ForceTest.exe throws this:

```
Enumerated Joystick : Arduino Due
Supported Effects : 
   - GUID_ConstantForce
   - GUID_RampForce
   - GUID_Square
   - GUID_Sine
   - GUID_Triangle
   - GUID_SawtoothUp
   - GUID_SawtoothDown
   - GUID_Spring
   - GUID_Damper
   - GUID_Inertia
   - GUID_Friction
   - GUID_CustomForce

Creating effects for Arduino Due
Error creating Spring Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
Error creating Friction Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
Error creating Damper Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
Error creating Constant Force Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
Error creating Periodic Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
Error creating Pavement Groove Effect. Your joystick may not support this effect.  : 
E_NOTIMPL : 
The function called is not supported at this time
```

I will try to have a look at it later...

billydragon, 2021-02-22:
ForceTest.exe is not big deal because it using some extra method (i think so) to get report about created effect. I will fix it later. Now Due will work with other tester like WheelCheck_1_72. I forget some thing. You need to modify some think in USB core to fix problem received data from USB. This bug is from DUE Core. 

My Yoke for MSFS2020 almost works fine except need to calibration for smooth moving axis.
here is complete project of my Yoke using 2 AC Servo motor driving and Due. Not perfect optimized because i am not Programmer guy, just for hobby. But it work. :-)
[https://github.com/billydragon/FFBYoke.git](url)
and configuration app:
[https://github.com/billydragon/FFBYokeConfig.git](url)

I try ported this lib to STM32F407 and get some result and will share to you guys when finished.

NCPlyn, 2021-02-23:
Yeah, I took a look at your repositories and tried it. ForceTest didn't threw any errors, but DUE only showed me that the forces[1] changed for few packets and then went to 0. WheelCheck worked nicely. But when I started Forza Horizon 4, I was not receiving any force data. I will try to look at it again...

billydragon, 2021-02-23:
You need change gains. My default gains is very low because of AC servo driver and setting.

From: NCPlyn
Sent: Tuesday, February 23, 2021 2:34 PM
To: YukMingLaw/ArduinoJoystickWithFFBLibrary
Cc: Nguyen Duong; Mention
Subject: Re: [YukMingLaw/ArduinoJoystickWithFFBLibrary] FFB support for Arduino DUE (32bit ARM) (#23)

Yeah, I took a look at your repositories and tried it. ForceTest didn't threw any errors, but DUE only showed me that the forces[1] changed for few packets and then went to 0. WheelCheck worked nicely. But when I started Forza Horizon 4, I was not receiving any force data. I will try to look at it again...

NCPlyn, 2021-02-23:
Interesting is that I get data from Assetto Corsa and Project Cars 2. But not with Forza Horizon 4.
Even more interesting is that I can get force data from FH4 if I use Leonardo and not DUE...

billydragon, 2021-02-23:
I talk you that DUE is not fished develop by Arduino. USBCore is so many bugs. Function of receive data is missing in fifo buffer. So I try make it work but not like Leonardo. Leonardo is small flash memory and periferial just simple job is done. Now I moving to stm32f407 base on this dinamicHID lib. The FFB is all function work. ADC, PWM, Fast timer 32bit encoder ready work. I still can’t make composite USB to make configuration software communicated with it. Still leaning and reseaching. If you have experience about USB HID develop, you can help me.

From: NCPlyn
Sent: Tuesday, February 23, 2021 11:17 PM
To: YukMingLaw/ArduinoJoystickWithFFBLibrary
Cc: Nguyen Duong; Mention
Subject: Re: [YukMingLaw/ArduinoJoystickWithFFBLibrary] FFB support for Arduino DUE (32bit ARM) (#23)

Interesting is that I get data from Assetto Corsa and Project Cars 2. But not with Forza Horizon 4.
Even more interesting is that I can get force data from FH4 if I use Leonardo and not DUE...

NCPlyn, 2021-02-23:
Ok, I see what you mean. I just started with HID stuff.. I will be closing this issue because like you said. The problem is more in the USBCore of the ArduinoCore-sam.

closed 2021-02-23 by NCPlyn