YukMingLaw/ArduinoJoystickWithFFBLibrary #29: How to connect output steppers - Its not an issue with librarym more like need help in using it

gaurav2827, 2021-04-16:
Thanks a lot for this wonderful library. Detailed work indeed!

I have gone through the "SingleJoystickFFB" in examples but unable to understand where or how to connect steppers in the arduino board. The sketch says pin 7, 6, 9 are outputs. Pin 9 gives PWM output (to motor driver) understood but unable to catch the idea of pin 6 & 7. I have used the previous Joystick library for my home cockpit building but I am new to controlling steppers.
Please help!

Regards,
Gaurav

gaurav2827, 2021-04-23:
Hi guys, anyone who can help me on this?

YukMingLaw, 2021-06-24:
Hi,
pin7,6 are used to determine the direction of motor rotation.

sgtnoodle, 2021-06-30:
Stepper motors require a special type of driver, and they commonly use two control pins, "direction" and "step". You need to precisely control the step signal to make the motor move. 

You can either use a pwm peripheral and vary the frequency, or you can use a timer interrupt to toggle it with code. You likely won't be able to do what you're hoping in the main loop as this joystick library without using pwm because the library takes several milliseconds per cycle to do its calculations and USB accesses.

gaurav2827, 2021-08-05:
> Hi,
> pin7,6 are used to determine the direction of motor rotation.

Thanks for this. I am trying to use the example code but unable to get any output from arduino. My motor simply wont run or respond to anything. Its detected on fsforce.exe but nothing after that. No input is registered in the application or the windows game controller window. 

If you have a working code, could you plz share in the example folder. 

Regards

sgtnoodle, 2021-08-05:
I wouldn't expect any sort of example to run out of the box without some amount of hardware specific tweaking. Here's my main sketch for a wheel with a DC motor driven by an H-bridge, and a SPI quadrature decoder IC. It uses a modified joystick library that I forked so I could fix some bugs and add default spring centering. You likely won't be able to even compile it, but you can see it as a reference of what all I had to do to get my wheel working.

```
#include <Joystick.h>
#include <LS7366.h>
#include <SPI.h>

#define ICR 600

#define DIR_PIN 8
#define PWM_PIN 9
#define BRAKE_PIN 10
#define LED_PIN 13

//#define COUNT_HALF_RANGE 61440  // 1.5 turn
#define COUNT_HALF_RANGE 51200  // 1.25 turn for 900 deg lock to lock

#define AXIS_HALF_RANGE 32767

#define ENABLE_JOYSTICK 1

#define ENABLE_SWEEP 0

// These were measured at about 12.3V
//#define MAX_RATE 35987.5f  // 1 turn
//#define MAX_RATE 23991.666666667f  // 1.5 turns
//#define MAX_RATE 28790.0f  // 1.25 turns

// This is at 14V.
// This number just worked out really nicely...
#define MAX_RATE 32767.0f  // 1.25 turns

#define HARDSTOP_DC (ICR * 0.8f)
#define HARDSTOP_FADE 1500.0f

#define FEEDBACK_DC (ICR * 0.25f)

// When applying a braking torque to the motor, there's a linear electrical power gain and a quadratic resistive power loss.
// When there's more electrical power than resistive loss, power gets dumped back into the supply, aka regen braking.
// If we knew the motor's and brake resistor's parameters, we could just calculate that power and compute the right
// average voltage to apply across the braking resistor. That's a lot of work, though, so instead we can just fudge it
// and measure voltage spikes with a multimeter. Too low, and the supply voltage will spike. Too high, and the resistor will
// get unnecessarily hot.
#define BRAKE_MULTIPLIER 0.5f

LS7366 myLS7366(7 /* chip select */);

class LowPassFilter
{
  public:
    LowPassFilter(float tc_sec)
      : _tc_sec(tc_sec),
        _tc_sec_reciprocal(1.0f/tc_sec)
    {}

    void reset(float output)
    {
      _output = output;
      _rate = 0.0f;
    }
    
    float output() const { return _output; }
    float rate() const { return _rate; }
    void update(float value, float dt_sec)
    {
      if (dt_sec >= _tc_sec)
      {
        _rate = (value - _output) / dt_sec;
      }
      else
[... 161 lines cut ...]
    if (fabsf(vel_filter.rate()) > max_accel)
    {
      max_accel = fabsf(vel_filter.rate());
    }
//    Serial.println(max_accel);

#if 0
    Serial.print("now: ");
    Serial.println(now);
    Serial.print("dt_ms: ");
    Serial.println(dt_sec * 1e3f);
    Serial.print("count: ");
    Serial.println(count);
    Serial.print("filtered pos: ");
    Serial.println(pos_filter.output());
    Serial.print("filtered rate: ");
    Serial.println(pos_filter.rate());
    Serial.print("axis_pos: ");
    Serial.println(axis_pos);
    Serial.print("force: ");
    Serial.println(forces[0]);
#endif
  }
}
```

gaurav2827, 2021-08-11:
Wow this is intense! Guys i just want a simple sketch to make my flight sim yoke auto centre. Nothing more. 

I have x and y axis for roll and pitch of flight stick, both connected to two dc motors with full h bridge. I can give pwm signal to motors and it moves within those commands if programmed without this library, i am just unable to use this library to auto centre my yoke using the signals from pots connected to the range of yoke. My yoke has physical end switches so no need to add a program for hard end stops...can someone plz help what i am missing from the example code. I used the same code to pass the DIRECTION and PWM to h bridge, but the motor simply doesn't move

sgtnoodle, 2021-08-11:
I don't think you need this force feedback library to implement motor based
centering. This library is specifically for implementing haptic feedback
from games. If literally all you want is hardcoded centering forces, you
can just use the standard Arduino joystick library. You still need to do
the motor control, and that will require writing some non-trivial code.

For one thing, you're going to need to model the "spring". The equation for
spring force is F= k*d, where k is the spring constant and d is the
distance the spring is stretched or compressed. You can measure d via your
potentiometer, and then just try different k values until it feels good.

That's going to give you an undamped spring. If you turn the yoke and let
go, it's going to oscillate back and forth. You can damp it. F = k*v, where
v is the velocity. You can compute that by taking the numerical derivative
of position, then once again tweak the constant until it feels good.

Note that you've just made what's called a PD controller, or a PID
controller without the I.

Doing all that will give you the appropriate force to apply on the motor.
Force, or torque in the case of a motor, is proportional to motor current.
PWMing an H-bridge doesn't control current, though, it controls applied
voltage. A spinning motor generates a voltage proportional to its velocity,
called back-EMF. The difference between the applied voltage and the
back-EMF causes current to flow through the motor, based on the motor's
resistance. In order to control force, you therefore need to know the
velocity of the motor (in order to know its back-EMF) and then feed that
back into the commanded H-bridge PWM.

If you don't do that, the motor will fight any sort of motion. Just try
powering up the H-bridge and setting the PWM to 0. The yoke should be very
hard to move. That's because the motion generates a voltage, and that
voltage subsequently generates a torque in the opposite direction, because
the H-bridge is applying 0 volts (equivalent to shorting the motor wires to
themselves). When the H-bridge is unpowered, the motor moves freely because
the FETs are all high impedance and so the voltage just floats around to
match the back EMF.

If you get all that working, then the yoke will probably feel pretty good.
If you're powering the motors with a power supply, though, you'll notice
that the centering force fades out whenever you move the yoke rapidly away
from centered. That's because when applying a motor torque that opposes
motion, kinetic energy is converted back to electric energy. That's called
regenerative braking. Power supplies are only designed to source current,
not sink current. When you dump current back into a power supply, its
output capacitors charge up and the supply voltage shoots up. When the
supply voltage shoots up, the applied voltage of the H-bridge also shoots
up. Regardless of what your microcontroller tries to do (within reason),
any attempt to "brake" the motor while it's moving will result in the
supply voltage shooting up rather than the motor slowing down. The typical
solution for this is to add what's called a brake resistor. The idea is
that the microcontroller connects a high power resistor to the supply
whenever the motor is regen braking. Rather than the current charging up
the output capacitors, it goes into the resistor instead and turns into
heat. Another solution would be to use a rechargeable battery rather than a
power supply, or use a fancy power supply designed to sink current. For my
wheel project, I used a 100W 2 ohm resistor and a PWMed FET.

If all you really want is centering spring forces, could you ditch the
motors and use actual springs instead? That would probably be a lot simpler.

anderatt, 2021-10-20:
Hello Guys,
What game do you recommend to test the Force Feedback Efects?

Is it library work well with Steam games, like Euro Truck Simulator?

Congratulation for this library!!