YukMingLaw/ArduinoJoystickWithFFBLibrary #36: Force direction is backwards

sgtnoodle, 2021-12-19:
It's possible I'm using the library wrong, but I've spent several hours trying to debug this.

I've noticed that, in games that use constant forces to do their own feedback, the direction of the force is in the wrong direction. This obviously makes the games unplayable. Spring forces, friction and damping all seem to work correctly though.

When I use the program "ForceTest" to inject a constant force, the direction is indeed backwards from what Microsoft's documentation says the convention is. https://docs.microsoft.com/en-us/previous-versions/windows/desktop/ee417536(v=vs.85)

A force of "60" with a direction of "90" should come from the direction of positive X, which on my steering wheel should push counterclockwise. I get a clockwise force instead, though.

Likewise, when I do a periodic force, the wheel is forced clockwise and then counterclockwise.

sgtnoodle, 2021-12-19:
I just tried changing the diection of angle_ratio:
```
    float angle_ratio = axis == 0 ? -sin(angle) : cos(angle);
```
And now the direction of forces seem consistent, and spring centering and damping still work. I'm guessing that's because the `useForceDirectionForConditionEffect ` bool happens to be false in my setup, and so the angle_ratio doesn't matter for them.

The Microsoft documentation says that an angle of 0 is the -y axis, and rotations are clockwise. That means that a rotation of 90 degrees should fall on the positive x axis. My tweak to angle ratio above therefore seems to make sense:

For the X axis:
Angle 90 -> -1 (Push in the negative x direction)
Angle 270-> 1 (push in the positive x direction)

For the Y axis:
Angle 0 -> 1 (Push in the positive y direction)
Angle 180 -> -1 (push in the negative y direction)

playerNul, 2022-02-07:
Hi, im having same problem. How and where i have to add the code you writed (float angle_ratio = axis == 0 ? -sin(angle) : cos(angle);)  in SingleJoystickFFB code which is provided by the YukMingLaw? I am using it with just pwm correction for avoiding dc motor noise which i found before. There is no change in that code because i am noob level on coding and sorry for may bad english :)) i hope i have explained my problem.

And are you sharing your code in anywhere? I would like to receive inspiration from your code too. Thanks a lot.

playerNul, 2022-02-14:
Hi again. I have found where to add it. Thanks anyway :))