YukMingLaw/ArduinoJoystickWithFFBLibrary #38: DiRT 2.0 problem

TomasTheCreato, 2022-02-07:
Hi! Thanks for a great repo! My code seams to work with wheelcheck, but i don get any force data from Dirt 2.0??

YukMingLaw, 2022-02-19:
I tested the lib in AC & Project-Cars, it works well. But in Horizon 4 it seems would slow down the fps in the game.I will the it in DiRT 2.0 soon. Due to my endless work it maybe need few weeks.

YukMingLaw, 2022-02-19:
I tested the lib in AC & Project-Cars, it works well. But in Horizon 4 it seems would slow down the fps in the game.I will test it in DiRT 2.0 soon. Due to my endless work it maybe need few weeks.

LaviMahal, 2022-04-01:
> I tested the lib in AC & Project-Cars, it works well. But in Horizon 4 it seems would slow down the fps in the game.I will test it in DiRT 2.0 soon. Due to my endless work it maybe need few weeks.

Any updates on this??

bogdan-dumitrescu, 2025-02-09:
For Dirt Rally 2.0, start the game as normal. After the game asks you to press a button or hit enter to start, disconnect your wheel's USB cable and plug it back in. The game will inform you that a device was reconnected and the FFB will work.