YukMingLaw/ArduinoJoystickWithFFBLibrary #39: how to work with this library

DimaMosur, 2022-02-08:
Hello everyone, I'm new to arduino and I'm trying to write my own steering wheel with FFB, so far it's not very good. I can write code without FFB but I want to have it I have everything I need to create it motor and h-bridges the code I try to write gives me a lot of errors that I don't know how to fix. I would be very grateful if there is somewhere a code with ffb on this library, because while I understand a little

DimaMosur, 2022-02-15:
addition

when I make my steering wheel on the original library, the code works for me and when I calibrate the device everything is correct, but when I do it on this library, the code is also compiled but does not react to HID-device at the same time, if I make a welcome through the "println" com port, you can see that the signals there are also they are working if you do it without ffb on the library with ffb the story is the same when using the original library everything works correctly I don’t know if it’s my problem or it’s the library just like I already mentioned I’m a beginner

code with ffb on this library but of course it doesn’t work, I don’t know why the device also doesn’t react on the PC

`// ENCODER WIRING
// Red : VCC = 5V
// Black: 0V = GND
// White: OUTA = Pin 2
// Green: OUTB = Pin 3
//////////////////////
//SHIFT WIRING
//Black: GND
//Red: Pin 8
//White: Pin 9
//////////////////////
//PEDALBOARD WIRING
//Yellow (throttle): GND
//Orange (throttle): 5V
//Black (throttle): pin A0
//Green (brake): GND
//White (brake): 5v
//Brown (brake): pin A1
//////////////////////
//BUTTON WIRING
//

#include <Joystick.h>

Joystick_ Joystick(0x12, JOYSTICK_TYPE_JOYSTICK,
  4, 0,                  // Button Count, Hat Switch Count
  true, true, true,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  true, true, false);  // No accelerator, brake, or steering

const int outA = 0;
const int outB = 1;
volatile long encoder = 0; // declare volatile since modified by interrupt routines
long encoder_save = 0;

Gains mygains[2];
EffectParams effects[2];
int32_t forces[2] = {0};

void setup() {
  Joystick.begin();
  Joystick.setXAxisRange(-1600, 1600);
  Joystick.setYAxisRange(0, 1023);
  Joystick.setZAxisRange(0, 1023);
  pinMode(8, INPUT_PULLUP); //Shift pin
  pinMode(9, INPUT_PULLUP); //Shift pin
  pinMode(outA, INPUT_PULLUP); //Encoder pin
  pinMode(outB, INPUT_PULLUP); //Encoder pin
  // attach interrupts to pins
  // call digitalPinToInterrupt(pin) to be compatible with different Arduino boards
  attachInterrupt(digitalPinToInterrupt(outA), outAChange, CHANGE);
  attachInterrupt(digitalPinToInterrupt(outB), outBChange, CHANGE);

  mygains[0].totalGain = 100;//0-100
  mygains[0].springGain = 100;//0-100
  mygains[0].damperGain = 100;
  mygains[0].inertiaGain = 100;
  mygains[0].frictionGain = 100;   

  Joystick.setGains(mygains);
}

void loop() {
[... 122 lines cut ...]
    Joystick.setButton(0, 1);
    } else {
     Joystick.setButton(0, 0);
    }

  //Shift Down
  if(digitalRead(9) == LOW){
    Joystick.setButton(1, 1);
    } else {
     Joystick.setButton(1, 0);
    }
    
  delay(10);        // delay in between reads for stability
}

// interrupt routines
void outAChange() {
  // when outA changes, outA==outB means negative direction
  encoder += digitalRead(outA) == digitalRead(outB) ? -1 : 1;
}

void outBChange() {
  // when outB changes, outA==outB means positive direction
  encoder += digitalRead(outA) == digitalRead(outB) ? 1 : -1;
}`

YukMingLaw, 2022-02-19:
which arduino hardware did you use for the project?

DimaMosur, 2022-02-20:
arduino leonardo

YukMingLaw, 2022-02-21:
```
- Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK, 4, 0,
+Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_MULTI_AXIS, 4, 0,
true, true, true, //X,Y,Z
true, true, true,//Rx,Ry,Rz
false, false, false, false, false);
```
try this code?

vtrx-az, 2022-03-01:
I'm trying to port this code to STM32F103 using keil compiler.
I made changes to behave like C code (C99 mode).
The build completed perfectly, but the effects do not follow the timing.
Using fedit.exe, the effects work as if they are in infinite time, they don't stop.
'Periodic Force' does not work.
I have checked the code several times and I don't understand where the error is.
How is the 'Periodic Force' effect controlled and what could be wrong with looping other effects?

1781741527, 2023-04-08:
Hello, I would like to ask, have you ever encountered a situation where the force feedback done by this library will cause the game to drop frames in the game? Or freeze and dodge? I also got a similar issue to you, I seem to have solved the problem of force feedback acquisition, but it causes my game to drop frames very badly!

Hello, I would like to ask, have you ever encountered a situation where the force feedback done by this library causes the game to drop frames in the game? I'm also having a similar issue, I seem to have fixed the issue with force feedback acquisition, but it's causing my game to drop frames very badly!
I feel like the device is constantly fetching game data from the game, causing the game to drop frames, but I don't know how to fix it yet, so I'm wondering if your project is complete? Can you give me some help?

DimaMosur, 2023-04-19:
I also had slow fps in most games, unfortunately I abandoned this project at the moment