YukMingLaw/ArduinoJoystickWithFFBLibrary #41: STM32F103

vtrx-az, 2022-03-01:
I'm trying to port this code to STM32F103 using keil compiler.
I made changes to behave like C code (C99 mode).
The build completed perfectly, but the effects do not follow the timing.
Using fedit.exe, the effects work as if they are in infinite time, they don't stop.
'Periodic Force' does not work.
I have checked the code several times and I don't understand where the error is.
How is the 'Periodic Force' effect controlled and what could be wrong with looping other effects?

vtrx-az, 2022-03-13:
Solved.

fikririzal, 2023-01-28:
can i ask about how you solve this problem?

does this library support STM32F103 / STM32F401?

vtrx-az, 2023-01-29:
> can i ask about how you solve this problem?
> 
> does this library support STM32F103 / STM32F401?

I abandoned the project because it was crashing a lot.
I changed the descriptor many times and sometimes I don't answer anymore.

mckelektronik, 2023-09-08:
Are there any developments about the F103?