YukMingLaw/ArduinoJoystickWithFFBLibrary #42: Y-axis work in MSFS2020 but not X-axis?!?!

crimzzon92, 2022-03-04:
So I make the code work and the y-axis works whit everything but not the x-axis. The x-axis and y-axis work in the force tester but in MSFS2020 only the Y-axis works correctly. The x-axis (roll) is banking to the left 100%. Y-axis (pitch) fins the center and works perfectly. any anyone knows way this is?

----------------My code---------------

#include "Joystick.h"

int32_t forces[2] = {0};

int PITCH_R_EN = 16;
int PITCH_L_EN = 16;
int PITCH_R_PWM = 10;
int PITCH_L_PWM = 5;

int ROLL_R_EN = 8;
int ROLL_L_EN = 8;
int ROLL_R_PWM = 9;
int ROLL_L_PWM = 6;

//X-axis & Y-axis REQUIRED
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK, 
  0, 0, //Button Count, Hat Switch Count
  true, true, false, //X,Y,Z
  false, false, false,//Rx,Ry,Rz
  false, false, false, false, false); //Rudder, Throttle, Accelerator, Brake, Steering

Gains mygains[2];
EffectParams myeffectparams[2];

void setup(){
//In & Outputs For Pitch Motor  
pinMode(PITCH_R_EN, OUTPUT);
pinMode(PITCH_L_EN, OUTPUT);
pinMode(PITCH_R_PWM, OUTPUT);
pinMode(PITCH_L_PWM, OUTPUT);
digitalWrite(PITCH_R_EN, HIGH);
digitalWrite(PITCH_L_EN, HIGH);
pinMode(A0,INPUT);

//In & Outputs For Roll Motor  
pinMode(ROLL_R_EN, OUTPUT);
pinMode(ROLL_L_EN, OUTPUT);
pinMode(ROLL_R_PWM, OUTPUT);
pinMode(ROLL_L_PWM, OUTPUT);
digitalWrite(ROLL_R_EN, HIGH);
digitalWrite(ROLL_L_EN, HIGH);
pinMode(A1,INPUT);
    
    Joystick.setXAxisRange(0, 1023);
    Joystick.setYAxisRange(0, 1023);
   
    //set X Axis gains
    mygains[0].totalGain = 30;//0-100
    mygains[0].springGain = 30;//0-100
    //mygains[0].damperGain = 10;//0-100
    //mygains[0].inertiaGain = 10;//0-100
    //mygains[0].frictionGain = 10;//0-100

    //set Y Axis gains
    mygains[1].totalGain = 10;//0-100
    mygains[1].springGain = 10;//0-100
    //mygains[1].damperGain = 10;//0-100
    //mygains[1].inertiaGain = 10;//0-100
    //mygains[1].frictionGain = 10;//0-100
    
    //enable gains REQUIRED
    Joystick.setGains(mygains);
    Joystick.begin();
}

void loop(){
  int valueX = analogRead(A1); //Pitch Input
  int valueY = analogRead(A0); //Roll Input
  
  //set X Axis Spring Effect Param
  myeffectparams[0].springMaxPosition = 1023;
  myeffectparams[0].springPosition = valueX;//0-1023

  //set Y Axis Spring Effect Param
  myeffectparams[1].springMaxPosition = 1023;
  myeffectparams[1].springPosition = valueY;//0-1023

  //set X Axis Damper Effect Param
  //myeffectparams[0].damperMaxVelocity = 1023;
  //myeffectparams[0].damperVelocity = valueX;//0-1023

  //set Y Axis Damper Effect Param
  //myeffectparams[1].damperMaxVelocity = 1023;
  //myeffectparams[1].damperVelocity = valueY;//0-1023

  //set X Axis Inertia Effect Param
  //myeffectparams[0].inertiaMaxAcceleration = 1023;
  //myeffectparams[0].inertiaAcceleration = valueX;//0-1023

  //set Y Axis Inertia Effect Param
  //myeffectparams[1].inertiaMaxAcceleration = 1023;
  //myeffectparams[1].inertiaAcceleration = valueY;//0-1023

  //set X Axis Friction Effect Param
  //myeffectparams[0].frictionMaxPositionChange = 1023;
  //myeffectparams[0].frictionPositionChange = valueX;//0-1023

  //set Y Axis Friction Effect Param
  //myeffectparams[1].frictionMaxPositionChange = 1023;
  //myeffectparams[1].frictionPositionChange = valueY;//0-1023

  //Send HID data to PC
  Joystick.setXAxis(valueX);
  Joystick.setYAxis(valueY);

  //Recv HID-PID data from PC and caculate forces
  Joystick.setEffectParams(myeffectparams);
  Joystick.getForce(forces);
  
        // X Axis Funktion
  if(forces[0] > 0){
    analogWrite(ROLL_L_PWM,abs(forces[0]));
    analogWrite(ROLL_R_PWM, 0);
    digitalWrite(ROLL_L_EN,HIGH);
  }else{
    analogWrite(ROLL_R_PWM,abs(forces[0]));
    analogWrite(ROLL_L_PWM, 0);
    digitalWrite(ROLL_R_EN,HIGH);
  }
    delay(10);
        // Y Axis Funktion
    if(forces[1] > 0){
    analogWrite(PITCH_L_PWM, 0);
    analogWrite(PITCH_R_PWM,abs(forces[1]));
    digitalWrite(PITCH_R_EN,HIGH);
  }else{
    analogWrite(PITCH_R_PWM, 0);
    analogWrite(PITCH_L_PWM,abs(forces[1]));
    digitalWrite(PITCH_L_EN,HIGH);
  }
  delay(10);
}

crimzzon92, 2022-03-06:
Got it to work. uses the weel setting for the pot range on the X-axis (-512 - 512) instead of the y axis range (0 - 1023). It works in MFSM2020 for me now.

closed 2022-03-06 by crimzzon92

pablofridman, 2024-08-09:
how do you reed the position of the yoke without encoders?