YukMingLaw/ArduinoJoystickWithFFBLibrary #45: Try to make FFB on Z Axis (Rudder)??

crimzzon92, 2022-03-23:
Hi! I am working on my full FFB Yoke and Pedals. I have made them on separate pro micro bords and used the X-Axis force for the Z-Axis (Rudder). It works, but when using the yoke and I bank. The ruder pedals move to. I use XPForce on MSFS2020. Is there anyone who knows how I might solve this??

Thanx!

YukMingLaw, 2022-07-14:
This library just support two force feedback axis currently.I try to make Z axis force feedback come true.But, now you can make two device to implatement 4 axis force feedback and try them in your game.ofcourse it just my envisage, maybe the game not support.

MatthewReed303, 2022-12-01:
Hi @crimzzon92 Did you ever get this working? Thanks

crimzzon92, 2022-12-01:
No i did not

MatthewReed303, 2022-12-01:
Hi @YukMingLaw Do you have any plans to add Z axis support for rudder/yaw? XPForce supports Yaw so this would be great option to have available. Or can you point me in the right direction to add this support? Thanks

Selbstdenker-CH, 2023-07-30:
@YukMingLaw Thank you for the wonderful work. It is thanks to passionate people like you that DIYers can DIY and Open Source projects exist. I share the point of view that adding Z axis support, or rudder pedals would set your project apart from commercial FFB. 

This might then get traction in the Flight Sim community. 
Even more so if you make a simplified install tutorial (pictures with arrows and link to the supported board). Like for a one button joystick. I remarqued that people are afraid of text. They want to do things, without thinking :( 

On another note, is there any way we could support you ?