YukMingLaw/ArduinoJoystickWithFFBLibrary #51: Games doesn't return any force feedback

akuladon, 2022-07-31:
Hello 
I made a program based on your lib
When i am using WheelCheck arduino gets information about forces and returns them to the serial port
However, when i try to get forces in games, it doesn't see it. it returns 0
Here is my code
`#include "Joystick.h"

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK,
0, 0,
true, false, false,
false, false, false,
false, false,false, false, false);

Gains mygains[1];
EffectParams myeffectparams[1];
int32_t forces[1];

unsigned long t,lt,dt,lastT; //time part
int x, lx, deltaX, dx,ldx, dx2; //differential part

void setup() {
  mygains[0].totalGain = 50;   //setting gains
  mygains[0].rampGain = 50;
  mygains[0].squareGain = 50;
  mygains[0].sineGain = 50;
  mygains[0].triangleGain = 50;
  mygains[0].sawtoothdownGain = 50;
  mygains[0].sawtoothupGain = 50;
  mygains[0].springGain = 80;
  mygains[0].damperGain = 50;
  mygains[0].inertiaGain = 50;
  mygains[0].frictionGain = 50;
  mygains[0].customGain = 50;

  
  Joystick.setGains(mygains);
  Joystick.begin();
  Joystick.setXAxisRange(-512, 512);
  Serial.begin(500000);
  lt=millis();
}

void loop() {
  lt = t; //time part
  t = millis();
  dt = t - lt;

  if(millis()-lastT>0){

  lastT=millis();

  lx=x; //last x 
  ldx=dx; // last x velocity
  x=analogRead(A0)-512; //x
  deltaX=x-lx;// delta x
  dx=deltaX/dt;//x velocity
  dx2=(dx-ldx)/dt;//x acceleration 
  
  myeffectparams[0].springMaxPosition = 512;
  myeffectparams[0].springPosition = x;

  myeffectparams[0].damperMaxVelocity = 100;
  myeffectparams[0].damperVelocity = dx;

  myeffectparams[0].inertiaMaxAcceleration = 5;
  myeffectparams[0].inertiaAcceleration = dx2;

  myeffectparams[0].frictionMaxPositionChange = 10;
  myeffectparams[0].frictionPositionChange = deltaX;
  

  
  Joystick.setXAxis(x);
  Joystick.sendState();
  Joystick.setEffectParams(myeffectparams);
  Joystick.getForce(forces);
  
  Serial.print(x);
  Serial.print(" ");
  Serial.println(forces[0]);
  }
  
}`
I think games just don't recognize it as a force feedback device, that can use effects(like spring, friction, ect) because while Wheel check is working, Force Test says "Your joystick may not support this effect" for each effect
Do you know how to fix that?
Thanks in advance

akuladon, 2022-08-02:
I tried this code in Beamng.Drive and ETS2 
In beamng i turned on ForceFeedBack checkbox but it didn't gave me any force feedback
Then I tried to use Steering axis, but it also didn't give any results
The only app that gives me force value is Wheel Check 1.72

akuladon, 2022-08-25:
I tried assetto corsa and it works
I have changed code a bit, so there are three axes(also added second axis ffb, but i am not using it)
However it still doesn't work in ETS2 and BeamNG.Drive :(

1781741527, 2023-04-08:
Hello, I would like to ask, have you ever encountered a situation where the force feedback done by this library will cause the game to drop frames in the game? Or freeze and dodge? I also got a similar issue to you, I seem to have solved the problem of force feedback acquisition, but it causes my game to drop frames very badly!

Here's what I set up in my code:
//X-axis & Y-axis REQUIRED
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 12, 0,
  true, true, false, //X,Y,Z
  false, false, false,//Rx,Ry,Rz
  false, false,
  true, true, false);

I feel like the device may be getting data into the game, causing the game to drop frames, but I don't know how to fix it yet, so I want to know if your project is complete? Can you give me some help?

akuladon, 2023-05-05:
Yes, i also have this problem
It crashes games/slows physics/drops frame rate
I think leonardo doesn't have enough computing power to compute trigonometry functions(which are used in code to make effects) fast enough
This library is based on hoantv's library "VNWheel". VNWheel uses teensy 4.0 600 MHz instead of leonardo and it is much faster
I would like to buy one, but right now it costs too much for me. So now my project is frozen. I think you should abandon leonardo and try Teensy.
While writing this, i checked the prices on aliexpress and found some offers that are half of the past price, so maybe it is the best moment to buy one.

YukMingLaw, 2024-01-08:
Indeed, the situation is caused by insufficient computational capacity of the chip, and it's also related to whether the game employs a synchronous mechanism for sending force feedback data. The v0.9.0 version performed well in 'Project Cars 2', but significant frame drops occurred in both 'Horizon 4' and 'Horizon 5'. I invested effort into pinpointing and resolving this issue, and thus, I recommend you try building your project with v1.0.0. Enjoy it!