YukMingLaw/ArduinoJoystickWithFFBLibrary #52: Games crashing on start, or in control setting menu (arduino pro micro + encoder)

Freyls, 2022-09-01:
`
#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_MULTI_AXIS, 0, 0,false, false, false, false, false, false, false, false, false, false, true);

#define ENC_A 2 //Енкодер фаза А
#define ENC_B 3 //Енкодер фаза В

int16_t X_POS = 0;  

void setup() { 
  Joystick.begin();
  Joystick.setSteeringRange(-1500, 1500);
  Joystick.setSteering(0);
  pinMode (ENC_A, INPUT_PULLUP);
  pinMode (ENC_B, INPUT_PULLUP);
   
  attachInterrupt(digitalPinToInterrupt(ENC_A), ENC_TIC, RISING);
} 

void loop() { 
  if(X_POS >= -1500 && X_POS <= 1500) {
    Joystick.setSteering(X_POS);
  }
}

void ENC_TIC ()
{
  bool State = digitalRead(ENC_B);  
  if(State == LOW) {
      X_POS++;
  }
  else {
    X_POS--;
  }
}
`
i have an arduino pro micro on ATMega 32u4 and 600 P/R encoder, but I don't understand why it not working, thanks in advance for the answer.

Freyls, 2022-09-01:
i tested it out in acc(crashing in control settings) and f122(crashing on start)

Freyls, 2022-09-01:
i test it in acc(crashing in options), and f122(crashing on start), but both working with keyboard

akuladon, 2022-09-06:
As i know you must use setEffectParams and getForce because the game is waiting for your controller and doesn't do anything until your controller gives feedback.
Also you must use JOYSTICK_TYPE_JOYSTICK because windows doesn't see multi axis as a gaming device

closed 2022-10-13 by Freyls