YukMingLaw/ArduinoJoystickWithFFBLibrary #53: Force Valsue reverses after hitting 255

JonnyNOS, 2022-09-28:
Hi, the force feedback works just fine, but if i turn the wheel really fast, the Force[0] value reverses back until it hits 0 and then into negative. It feels like a Variable overshoots and goes back down instead of having a limit.

In my code i just get the Force Value and tell the wheel to turn in the direction depending on positive or negative and how high the value is.

This is my setup:
________________________________________________________

//Setup
void setup() {

  define_can_messages();
  pinMode(A3, INPUT);
  pinMode(A5, INPUT);
  Serial.begin(115200);

    Joystick.begin();
    //Steering wheel
    
    Joystick.setXAxisRange(0, 1023);
    //set X Axis gains
    mygains[0].totalGain = 100;//0-100                                                      //setting up everything
    mygains[0].springGain = 100;//0-100
    //mygains[0].damperGain = 100;
    //mygains[0].inertiaGain = 10;
    //enable gains REQUIRED
    Joystick.setGains(mygains);

  
  Joystick.begin();
  Setup_MCP2515();
  c = 0;
  
  }

___________________________________________________

And this is the Signal for the Steering wheel:
____________________________________________________

 //Lenkrad
void Signal_Lenkrad() {
  int HighP = canMsg.data[1];
  int LowP = canMsg.data[0];
  uint16_t  Position = (HighP << 8) + (LowP & 0xff);
  long corrPos = 0;
  corrPos = Position;
  if (canMsg.data[3] == 255) {
    corrPos = corrPos - 65535;
  } 

  Signalrdy = corrPos / 124 + 512;   // Wandelt corrPos in Signal von 0 bis 1023 um

    if (Signalrdy < 0) {          //
      Signalrdy = 0;              //
    }                             // Setzen des Maximums des Lenkeinschlags auf 0 - 1023
    else if (Signalrdy > 1023) {  //
      Signalrdy = 1023;           //
    }                             //

  int posChange = feedback + 255;

    
  //set EffectParameters for force feedback
  myeffectparams[0].springMaxPosition = 1023;
  myeffectparams[0].springPosition = Signalrdy; //0 - 1023

 // myeffectparams[0].damperMaxVelocity = 0;
 // myeffectparams[0].damperVelocity = 0;

  //myeffectparams[0].inertiaMaxAcceleration = 512;
  //myeffectparams[0].inertiaAcceleration = posChange;

  //myeffectparams[0].frictionMaxPositionChange = 255;
  //myeffectparams[0].frictionPositionChange = posChange;

Serial.println(feedback);

  
  //Joystick.getGains();
  Joystick.setEffectParams(myeffectparams);    
  Joystick.getForce(forces);                  //get the forces of the Wheel
  Joystick.setXAxis(Signalrdy);

}
_________________________________________________

After that i just insert the forces[0] value into the motor. but the forces[0]value itself is overshooting like written above.

JonnyNOS, 2022-09-29:
I found something in the Joystick.h file all the way at the bottom.

Mayby its something which could be fixed with the Limitation.

I mean right around this line:        //it should be added some limition here,but im so tired,it's 2:24 A.M now!

________________________

	//set gain functions
	int16_t setGains(Gains* _gains){
	    if(_gains != nullptr){

			//it should be added some limition here,but im so tired,it's 2:24 A.M now!
		
	        m_gains = _gains;

	        return 0;
	    }
	    return -1;
	};
	//set effect params funtions
	int16_t setEffectParams(EffectParams* _effect_params){
	    if(_effect_params != nullptr){
	        m_effect_params = _effect_params;
	        return 0;
	    }
	    return -1;
	};
};

YukMingLaw, 2022-10-15:
I analyze the reasonableness of adding output value restrictions to the code in the library, thank you for your feedback.

JonnyNOS, 2022-12-05:
I found out it works fine.
But in Assetto Corsa when the car starts to understeer, the forces(0) value goes down.
I took the check from "enhance understeer effect" and its way better now.

I would need to get JUST the "understeer effect" seperately so i can add it to my motor to make it harder to steer when the car understeers.
Is there any way to get these values out of the library?

Another thing would be my steering wheel being too snappy on low force(0) values, so it would be optimal if i can change the gain only for low values. But thats mainly a problem by the steering wheel itself

1781741527, 2023-04-08:
Hello, I would like to ask, have you ever encountered a situation where the force feedback done by this library will cause the game to drop frames in the game? Or freeze and dodge?

JonnyNOS, 2025-04-15:
Hi, maybe its a very late response now, but i found myself stumbling across here after a long time.

in my case i had less FPS in Assetto Corsa but halfing the ffb refresh rate in game was a fix.

JonnyNOS, 2025-05-06:
And to close this issue, it was a setting in assetto corsa which enhances the understeering effect.
Deactivating this, the force only reduced a bit when understeer occurred.

closed 2025-05-06 by JonnyNOS