YukMingLaw/ArduinoJoystickWithFFBLibrary #56: it causes my game to crash or get stuck directly，I need help！

1781741527, 2023-04-07:
Hello, I tried to use your library to get the force feedback information sent back from the game, but when I upload your routine to my Arduino Leonardo board, I find that it causes my game to crash or get stuck directly, and the game returns to normal when I reset or unplug the USB cable, so I wonder what is the cause of this phenomenon? How can it be improved? Would love to hear from you!

1781741527, 2023-04-08:
I feel like the device is constantly fetching game data from the game, causing the game to drop frames, but I don't know how to fix it yet, so I'm wondering if your project is complete? Can you give me some help?

Valdemir-DSW, 2023-05-24:
I am suffering from the same problem in (wheel force feedback test) it works normally but when I enter any game everything crashes I don't know how we can solve it

Valdemir-DSW, 2023-05-24:
My code might have some errors because I deleted some data that I couldn't show about my project but it's not part of the force feedback.

#include "BTS7960.h"

#include "Joystick.h"
#include <Encoder.h>
#define PINO_ENCODER_A 0
#define PINO_ENCODER_B 1

//X-axis & Y-axis REQUIRED
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_GAMEPAD, 24, 0,
  true, false, true, //X,Y,Z
  true, true, true,//Rx,Ry,Rz
  false, false, false, false, false);

  int gain = 60;
  int damper = 40;
  int inertia = 50;
  int friction = 44;
  int spring = 44;
  int constant = 65;

  int min_acel = 10;
  int max_acel = 1000;

  int min_frei = 10;
  int max_frei = 1000;

  int min_enbre = 10;
  int max_enbre = 1000;

  int min_hb = 10;
  int max_hb = 1000;
  

  

 
  int ppr = 600;

  String command;
int angle, decimal;

const int encoderPinA = 0;       // Trocando o pino A do encoder para o pino digital 1
const int encoderPinB = 1;       // Trocando o pino B do encoder para o pino digital 0
       
int angleMax = 900;        // Ângulo máximo em graus

Encoder encoder(encoderPinA, encoderPinB);
int angles = 0;
  

const uint8_t EN = 8;
const uint8_t L_PWM = 9;
const uint8_t R_PWM = 10;

BTS7960 motorffb(EN, L_PWM, R_PWM);

Gains mygains[2];
EffectParams myeffectparams[2];
int32_t forces[2] = {0};
int address = 0; 

void setup(){
 TCCR1B &= ~(_BV(CS12) | _BV(CS11) | _BV(CS10)); // Limpar os bits de prescaler existentes
 TCCR1B |= _BV(CS10); // Configurar o prescaler para 1    pwm  a 20 k

  pinMode(13, OUTPUT);

 

  Joystick.setRzAxisRange(min_hb, max_hb);
  Joystick.setRxAxisRange (min_frei, max_frei);
  Joystick.setXAxisRange (0, 1430);
  Joystick.setRyAxisRange(min_enbre, max_enbre);
  Joystick.setZAxisRange (min_acel, max_acel);

    
   
    
    //Steering wheel
    //Joystick.setXAxisRange(-512, 512);
    //set X Axis gains
    mygains[0].totalGain = gain;//0-100
    mygains[0].springGain = spring;//0-100
    mygains[0].constantGain = constant;//0-100
    mygains[0].frictionGain = friction;//0-100
    mygains[0].inertiaGain = inertia;//0-100
    mygains[0].damperGain = damper;//0-100
    
    //enable gains REQUIRED
    Joystick.setGains(mygains);
    Joystick.begin();

    // Inicialização do encoder
  encoder.write(0);
  EEPROM.begin();
  loadParameters();

}

void loop(){

  int encoderValue = encoder.read();
  angle = map(encoderValue, 0, ppr, 0, 360);
  int voltare = map(angle, 0, angleMax, 0, 360);
 // int valor360 = map(voltare, 0, angleMax, 0, 360);

  Joystick.setXAxis(voltare);
  

  motorffb.Enable();
  

Joystick.setRyAxis(analogRead(A2));
Joystick.setRxAxis (analogRead(A1));
Joystick.setRzAxis (analogRead(A3));
Joystick.setZAxis (analogRead(A0));
  while (Serial.available() > 0) {
/
/
/
  

  
  //Steering wheel
 myeffectparams[0].springMaxPosition = 1430;
 myeffectparams[0].springPosition = voltare  - 715; //-512-512

  //Send HID data to PC
  

  //Recv HID-PID data from PC and caculate forces
  //Steering wheel
  
  Joystick.setEffectParams(myeffectparams);
  Joystick.getForce(forces);
  if (forces[0] > 0) {
    motorffb.TurnRight(abs(forces[0]));
  } else {
    motorffb.TurnLeft(abs(forces[0]));
  }
}

YukMingLaw, 2024-01-08:
This issue should have been fixed in version v1.0.0, so you can try that out.