YukMingLaw/ArduinoJoystickWithFFBLibrary #58: FFB signal

Corbaax, 2023-07-09:
Hi,
I'm currently into a develpment of a custom FFB wheel firmware.
I really like your library but i'm having and issue whenever i turn on the FFB setting.
Seems like the game Is going on thrashing from all the call for FFB data and the physical engine of the game Is going like 200% of usage.

I think i'm forgetting something in the code like and interrupt or something like this,
Or else the Arduino Leonardo Is Just not enough for FFB calc.
I'm using an official Leonardo as hardware of this can help.

Thank you for your time and help

YukMingLaw, 2024-01-08:
Hello, thank you for your appreciation of this library. The issue you encountered might have been fixed in version v1.0.0, so it's worth trying to build your project using v1.0.0.

Corbaax, 2024-03-30:
hi, i really appreciate your work!
The 1.0.0 version works really fine.